luismesas/pydobot #4: No Port found for Dobot Magician

grappyX, 2017-11-30:
Hey Luismesas,

i've installed all required files to my raspberry pi 3, but as soon as i start the sample.py, ig get this error-message.

Can you help me?

morrisonj, 2017-12-02:
Did you configure the correct port? Should be something like the following on a Raspberry Pi:

```device = Dobot(port='/dev/ttyUSB0')```

Run ```ls /dev/tty*``` to try and find which is the correct USB port if it's not ttyUSB0.

grappyX, 2017-12-03:
I will try that, thanks for your fast response!!

luismesas, 2019-06-22:
Hi @grappyX ,

I updated the basic example to use the first available port, please let me know if this helps.

PS: I know I'm replying a year late, but I guess I owe you an answer :)

luismesas, 2019-06-22:
@morrisonj Thank you so much for helping others with this library :)

closed 2019-06-22 by luismesas