luismesas/pydobot #6: When i try to move dobot it gives me an output

Giulio2002, 2018-01-12:
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 52, in run
    self._get_pose()
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 89, in _get_pose
    response = self._send_command(msg)
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 65, in _send_command
    self._send_message(msg)
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 74, in _send_message
    self.ser.write(msg.bytes())
  File "/usr/local/lib/python2.7/dist-packages/pydobot/message.py", line 36, in bytes
    self.refresh()
  File "/usr/local/lib/python2.7/dist-packages/pydobot/message.py", line 29, in refresh
    self.checksum = self.checksum + int(self.params[i])
ValueError: invalid literal for int() with base 10: '['
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 52, in run
    self._get_pose()
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 89, in _get_pose
    response = self._send_command(msg)
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 65, in _send_command
    self._send_message(msg)
  File "/usr/local/lib/python2.7/dist-packages/pydobot/dobot.py", line 74, in _send_message
    self.ser.write(msg.bytes())
  File "/usr/local/lib/python2.7/dist-packages/pydobot/message.py", line 36, in bytes
    self.refresh()
  File "/usr/local/lib/python2.7/dist-packages/pydobot/message.py", line 29, in refresh
    self.checksum += self.params[i]
TypeError: unsupported operand type(s) for +=: 'int' and 'str'

morrisonj, 2018-01-13:
Off the top of my head I think this relates to you using Python 2.7. Try Python 3.x.

Giulio2002, 2018-01-13:
I will

closed 2018-01-13 by Giulio2002