luismesas/pydobot #12: device.suck(1) doesn't turn on suction cup

msnederland, 2019-04-15:
I am trying to get the suction cup to turn on and off but de "device.suck(1) or device.suck(0)" has no effect. The suction cup does work in Dobot studio.

Here's my code:

```
import time
from glob import glob

from pydobot import Dobot

available_ports = glob('/dev/cu*usb*')  # mask for OSX Dobot port
if len(available_ports) == 0:
    print('no port found for Dobot Magician')
    exit(1)

device = Dobot(port=available_ports[0])

device.speed(300)

for i in range(1):
    device.suck(1)
    device.suck(0)
    
time.sleep(1)

device.close()
```

What am I doing wrong?

morrisonj, 2019-04-15:
I had this issue too but a while ago so forgive me if I'm wrong but try changing line 148 of dobot.py from:
`        msg.id = 62`
to:
`        msg.id = 63`

luismesas, 2019-06-22:
Hi,

I checked with last protocol version and this is working correctly with the message id 62.

Reading into the documentation of protocol, seems like 62 is the SET operation and id 63 is GET operation.

Please let me know if this is still happening

closed 2019-06-22 by luismesas