luismesas/pydobot #18: Cant run basic.py

asifma, 2019-08-13:
Hi, I'm not an expert on python. I get this error message, does it say you anything?
Am I missing any pre-req? I installed the latest Dobot Studio.
Running this on my PC.

C:\Users\asif.mithawala\Desktop\pydobot>python basic.py
pydobot: COM3 open
pydobot: >> AA AA:2:240:1::15
pydobot: >> AA AA:2:245:1::10
pydobot: >> AA AA:34:80:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:85
pydobot: >> AA AA:18:81:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:128
pydobot: >> AA AA:10:82:3:00 00 20 41 00 00 48 43:191
pydobot: >> AA AA:10:83:3:00 00 C8 42 00 00 C8 42:150
pydobot: >> AA AA:2:10:0::246
Traceback (most recent call last):
  File "basic.py", line 6, in <module>
    device = Dobot(port=port, verbose=True)
  File "C:\Users\asif.mithawala\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pydobot\dobot.py", line 46, in __init__
    self._get_pose()
  File "C:\Users\asif.mithawala\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pydobot\dobot.py", line 102, in _get_pose
    self.x = struct.unpack_from('f', response.params, 0)[0]
AttributeError: 'NoneType' object has no attribute 'params'

s-gv, 2020-01-28:
I'm getting the same error: AttributeError: 'NoneType' object has no attribute 'params'

```
pydobot: COM5 open
pydobot: >> AA AA:2:240:1::15
pydobot: >> AA AA:2:245:1::10
pydobot: >> AA AA:34:80:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:85
pydobot: >> AA AA:18:81:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:128
pydobot: >> AA AA:10:82:3:00 00 20 41 00 00 48 43:191
pydobot: >> AA AA:10:83:3:00 00 C8 42 00 00 C8 42:150
pydobot: >> AA AA:2:10:0::246
Traceback (most recent call last):
  File "test.py", line 7, in <module>
    device = Dobot(port=port, verbose=True)
  File "C:\Users\sagar\AppData\Local\Programs\Python\Python36\lib\site-packages\pydobot\dobot.py", line 46, in __init__
    self._get_pose()
  File "C:\Users\sagar\AppData\Local\Programs\Python\Python36\lib\site-packages\pydobot\dobot.py", line 102, in _get_pose
    self.x = struct.unpack_from('f', response.params, 0)[0]
AttributeError: 'NoneType' object has no attribute 'params'
```

s-gv, 2020-01-28:
I'm using the Dobot Magician with the firmware V 3.6.15.

s-gv, 2020-01-28:
It works fine in Linux but not in Windows. 

> It seems like I have the same error, don't know if I have installed everything correctly. Though I have a Macbook instead. Also when I run the example script from this repo in DobotStudio it returns "No module named pyDobot".
> /usr/local/opt/python/bin/python3.7 /Users/petter/Documents/BTH/Programvaruutveckling/src/dobot_script.py
> pydobot: /dev/cu.Bluetooth-Incoming-Port open
> pydobot: >> AA AA:2:240:1::15
> pydobot: >> AA AA:2:245:1::10
> pydobot: >> AA AA:34:80:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:85
> pydobot: >> AA AA:18:81:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:128
> pydobot: >> AA AA:10:82:3:00 00 20 41 00 00 48 43:191
> pydobot: >> AA AA:10:83:3:00 00 C8 42 00 00 C8 42:150
> pydobot: >> AA AA:2:10:0::246
> Traceback (most recent call last):
> File "/Users/petter/Documents/BTH/Programvaruutveckling/src/dobot_script.py", line 5, in
> device = Dobot(port=port, verbose=True) # creates a object of device
> File "/usr/local/lib/python3.7/site-packages/pydobot/dobot.py", line 46, in **init**
> self._get_pose()
> File "/usr/local/lib/python3.7/site-packages/pydobot/dobot.py", line 102, in _get_pose
> self.x = struct.unpack_from('f', response.params, 0)[0]
> AttributeError: 'NoneType' object has no attribute 'params'

> I'm getting the same error: AttributeError: 'NoneType' object has no attribute 'params'
>
> ```
> pydobot: COM5 open
> pydobot: >> AA AA:2:240:1::15
> pydobot: >> AA AA:2:245:1::10
> pydobot: >> AA [...]

It seems to work fine in Linux (Ubuntu 16.04) but the error shows up in Windows.

sarohm, 2020-05-04:
I have installed the library but getting execution errors:
ModuleNotFoundError: No module named 'message'

Nier-Ace, 2020-05-07:
@sarohm I have the same Problem:

Traceback (most recent call last):
  File "/home/ubuntu-laptop/PycharmProjects/DobotTest/DobotTestPy.py", line 3, in <module>
    from pydobot import Dobot
  File "/home/ubuntu-laptop/PycharmProjects/DobotTest/venv/lib/python3.6/site-packages/pydobot/__init__.py", line 1, in <module>
    from pydobot.dobot import Dobot
  File "/home/ubuntu-laptop/PycharmProjects/DobotTest/venv/lib/python3.6/site-packages/pydobot/dobot.py", line 7, in <module>
    from message import Message
ModuleNotFoundError: No module named 'message'

Did you manage to solve this?
(I was testing this library with the example code.)

gioargyr, 2020-05-07:
I am -more or less- in the same situation.
I am trying to install the module named "message" and it needs the module "observable".
The "observable" is installed successfully, but the "message" still has problems.
In my case, I get the error  "AttributeError: module 'message' has no attribute '__all__' " .
For all python-module installations I am referring to, I did them with pip 20.1.

I really have no solutions right now. I don't know what's going on with the "message" module.

Nier-Ace, 2020-05-14:
I tried to install the "message" library and installed the "observable" module after message has failed.
And i got the same error as you.

I tried to install both packages **via terminal** and "message" did successfully install.
The problem is, i'm using Pycharm as my IDE and it looks like Pycharm isnt using those installed Librarys via terminal. Which means there is a problem with my IDE and these packages and i dont know how to fix this. Do you know if there is a possibility to tell Pycharm to use those installed Librarys instead of the Pycharm Library?

Nier-Ace, 2020-05-14:
I might have found the issue.
It looks like my System Interpreter is using Python 2.7 and my Project is 3.6.
If i'm installing this via console it should be using the 2.7 Interpreter while installing this in my Project environment its 3.6...

gioargyr, 2020-05-14:
Hello Nier-Ace, what is your machine's OS? Windows, Linux or something else?
I am trying to understand how you successfully installed the "message" module.

As for the 2 different Pythons, I may have some advice, so if you want send me an e-mail. It's surely a Pycharm issue and we better not spam this topic with such an issue.

joygetsit, 2021-01-22:
For me, the error was due what roarandrock said. It picked the wrong port.
Changing the code from 'port = list_ports.comports()[0].device' to 'port = list_ports.comports()[1].device' 
did the trick for me.

luismesas, 2021-03-05:
> For me, the error was due what roarandrock said. It picked the wrong port.
> Changing the code from 'port = list_ports.comports()[0].device' to 'port = list_ports.comports()[1].device'
> did the trick for me.

I'm adding a print of all available ports in the basic example so you can see what port applies in your case

luismesas, 2021-03-05:
Basic example now works on version 1.3.0

closed 2021-03-05 by luismesas