luismesas/pydobot #32: pydobot installation

gioargyr, 2020-05-07:
Hello,

I am trying to install pydobot with "pip install pydobot" command.
It seems that there are some prerequisites.
I get errors and I already installed the packages wheel, observable and I cannot install the package message.

I have Windows 10 64-bit and Python 3.8.1 64-bit

gioargyr, 2020-05-14:
Hello again,

I tried again to install your module. I kept bagging my head on the "message" module, but as it seems this module works only for Python 2.7
So, this time I worked in a Linux machine (ubuntu 18.04) and used Python 2.7 and simple pip (not pip3).
It seems that things are better as I managed to install the following modules:
message, serial, enums, ptpy

Yet again, while installing pydobot, I get an error because you define in some point "from enums.PTPMode import PTPMode" and in that point the system gives 
"ImportError: No module named PTPMode"

So, I am more than willing to use your module, but I would like you to be a bit more precise on which platform it works and which are the prerequisites.

Hope I will hear from you soon.

Nier-Ace, 2020-05-16:
Hello there. Its me again!
I searched a little bit and someone with the same issue came up with a solution.
I trying this myself now.
But in case i'm too stupid here's the link: https://github.com/luismesas/pydobot/issues/31

Wish you luck.

Nier-Ace, 2020-05-16:
I tried this with python 2.7 and python 3.6 but it didn't work for me.
With 2.7 i get: ImportError: No module named PTPMode
I changed the code in dobot.py to:

import serial
import struct
import time
import threading
import warnings

from .message import Message
from .enums.PTPMode import PTPMode
from .enums.CommunicationProtocolIDs import CommunicationProtocolIDs
from .enums.ControlValues import ControlValues
...

And with 3.6 i get:

ModuleNotFoundError: No module named 'pydobot.enums'

-> with the same changes

tlack, 2020-08-03:
If you guys are still having problems, I was able to get it running on Python 3.8.5 with some hacks. Check out my fork here: https://github.com/tlack/pydobot

Crazy4Frog, 2020-09-17:
Hello. I worked with tlack's repository and found several errors in dobot.py and enums.
[pydobot.zip](https://github.com/luismesas/pydobot/files/5237191/pydobot.zip) There is fixed on 17.09.2020. I just added .value after PTPMode, CommunicationProtocolIDs and ControlValues's arguments and in CommunicationProtocolIDs corrected SET_GET_PTP_COORDINATEP_ARAMS to SET_GET_PTP_COORDINATE_PARAMS (Error is P in parrams).
I spent 4 hours for that and was happy when it worked. Thank for @tlack, but I hope you'll fix this in your repository

pgsell, 2020-12-13:
Thanks Crazy4Frog!
Found your message after 2 hours of debugging.. hopefully this fix gets included for others!

luismesas, 2021-03-05:
thanks guys for this thread, I'm working right now on including this on the main branch

luismesas, 2021-03-05:
Hi, I have integrated the changes of @tlack, if you update to version 1.3.0 it should be ok now.

Thanks!

closed 2021-03-05 by luismesas