luismesas/pydobot #33: dobot is connect with windows laptop

saransarval, 2020-08-20:
pydobot: COM3 open
pydobot: >> AAAA:18:81:3:00004843000048430000484300004843:128
pydobot: >> AAAA:10:83:3:0000484300004843:148
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\python3\lib\site-packages\pydobot\dobot.py", line 47, in __init__
    self._set_ptp_common_params(velocity=200.0, acceleration=200.0)
  File "C:\python3\lib\site-packages\pydobot\dobot.py", line 132, in _set_ptp_common_params
    return self._send_command(msg)
  File "C:\python3\lib\site-packages\pydobot\dobot.py", line 66, in _send_command
    response = self._read_message()
  File "C:\python3\lib\site-packages\pydobot\dobot.py", line 80, in _read_message
    msg = Message(b)
  File "C:\python3\lib\site-packages\pydobot\message.py", line 11, in __init__
    self.len = b[2]
IndexError: index out of range
>>>

luismesas, 2021-03-05:
is this exception happening when using the basic example? can you please update to version 1.3.0 (just released) and confirm if it still happens? thanks

MikeMikeHHN, 2021-11-17:
Issue still occures as of now, using the latest release.

jKostet, 2021-12-02:
Hi, I'm also trying to get this working on windows and have the same error. Just installed pydoboy with pip install, I'm using python 3.7. Dobot connects fine to Dobot Studio but I'd prefer using just python. Any support much appreciated 😊

jonascj, 2021-12-02:
@jKostet  Please report you version of windows, version of pydobot (latest probably), the program which fails to function (connect to the robot) and the error message you see (the same as saravanakumarBalakrishnan?).

jKostet, 2021-12-03:
@jonascj Windows 10 Enterprise, latest pydobot latest / 1.3.2. I'm trying to run the example in the README.

In addition I have the Dobot DLLs from the latest official Dobot 64bit python demo copied to where my python.exe is located.

Here is the full error msg:

`& 'C:\Program Files\Python37\python.exe' .\example.py

available ports: ['COM5', 'COM3']

pydobot: COM5 open

pydobot: >> AA AA:2:240:1::15

pydobot: >> AA AA:2:245:1::10

Traceback (most recent call last):

  File ".\example.py", line 9, in <module>

    device = pydobot.Dobot(port=port, verbose=True)

  File "C:\Users\USERNAME\AppData\Roaming\Python\Python37\site-packages\pydobot\dobot.py", line 31, in __init__

    self._set_queued_cmd_clear()

  File "C:\Users\USERNAME\AppData\Roaming\Python\Python37\site-packages\pydobot\dobot.py", line 241, in _set_queued_cmd_clear

    return self._send_command(msg)

  File "C:\Users\USERNAME\AppData\Roaming\Python\Python37\site-packages\pydobot\dobot.py", line 89, in _send_command

    response = self._read_message()

  File "C:\Users\USERNAME\AppData\Roaming\Python\Python37\site-packages\pydobot\dobot.py", line 80, in _read_message

    msg = Message(b)

  File "C:\Users\USERNAME\AppData\Roaming\Python\Python37\site-packages\pydobot\message.py", line 11, in __init__

    self.len = b[2]

IndexError: index out of range`

jonascj, 2021-12-03:
@jKostet  Thank you. I've just tried the example from the readme on Windows 10 Home and it works without installing Dobot Studio and without installing additional drivers[1]. This library does not communicate with the Dobot via the DLL-files as the official Python example, this library communicates with the Dobot directly over the serial-over-usb connection.

I notice that you have two available COM-ports, COM3 and COM5. Could it be that you (the readme example really) are using the wrong port?

You could try this change to the example:
```
port = available_ports[1].device # Changed from avilable_ports[0].device
```

Or try this instead, it lets you choose COM-port:

```
available_ports = list_ports.comports()
print('Available COM-ports:')
for i, port in enumerate(available_ports):
    print(f"  {i}: {port.description}")

choice = int(input('Choose port by typing a number followed by [Enter]: '))

port = available_ports[choice].device
```

[1] https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers#software This driver might be necessary, if Windows does not have a serial-over-usb driver installed already.

jonascj, 2021-12-03:
@MikeMikeHHN Thank you for reporting the issue as well. Would you provide your Windows version, pydobot version (1.3.2 probably), the python program which fails (the readme example?) and the exact error message you are seeing?

jKostet, 2021-12-06:
@jonascj COM5 is what I use with Dobot Studio. Also if the Dobot Studio is connected to the robot, pydobot will give the appropriate error, something like "Dobot is reserved". I'll try COM3 as well, sometimes the ports work a bit weird.

I got pydobot working on a raspberry pi, which will be the actual device I'll be running the Dobot from, so that's good. Getting it working on Windows would be great for easier development though. 😃

MikeMikeHHN, 2021-12-06:
@jonascj 
Issue arrises under Win10 Home 10.0.19042 Build 19042, pydobot 1.3.2 in the readme example aswell as a selfwritten program that uses pydobot. Error message:
`
available ports: ['COM4']
Traceback (most recent call last):
  File "C:\Users\Hati\Documents\GitHub\pydobot\basic_example.py", line 9, in <module>
    device = pydobot.Dobot(port=port, verbose=False)
  File "C:\Users\Hati\Documents\GitHub\pydobot\pydobot\dobot.py", line 31, in __init__
    self._set_queued_cmd_clear()
  File "C:\Users\Hati\Documents\GitHub\pydobot\pydobot\dobot.py", line 244, in _set_queued_cmd_clear
    return self._send_command(msg)
  File "C:\Users\Hati\Documents\GitHub\pydobot\pydobot\dobot.py", line 92, in _send_command
    response = self._read_message()
  File "C:\Users\Hati\Documents\GitHub\pydobot\pydobot\dobot.py", line 83, in _read_message
    msg = Message(b)
  File "C:\Users\Hati\Documents\GitHub\pydobot\pydobot\message.py", line 11, in __init__
    self.len = b[2]
IndexError: index out of range`

Anything else I can do to help?
On the same note, there is a bad workaround I found, usually flashing the firmware again aswell as reseting via reset switch results in the error not occuring for a while. I know, not a good way but its what I got so far ^^