luismesas/pydobot #34: Error When Running the Demo

Jeffery-Zhou, 2020-11-24:
I used python3 to run the basic code, the code returned with "No module 'message'".
------------------ ENV ---------------------- 
Ubuntu 1804,
Python 3.6, python 3.8
------------------ Log ---------------------- 
Traceback (most recent call last):
  File "/home/jeffery-hw/Desktop/MagneticManipulation/DobotController.py", line 3, in <module>
    from pydobot import Dobot
  File "/home/jeffery-hw/anaconda3/lib/python3.8/site-packages/pydobot/__init__.py", line 1, in <module>
    from pydobot.dobot import Dobot
  File "/home/jeffery-hw/anaconda3/lib/python3.8/site-packages/pydobot/dobot.py", line 7, in <module>
    from message import Message
ModuleNotFoundError: No module named 'message'

Then, I installed the message package and returned error as below:
I checked some answers on StackOverflow: it said I have some self-defined python filed called message, but actually my env is pure and I just reinstall my ubuntu system.
-----------------------------------------------------------------------------------------------------------------
    ERROR: Command errored out with exit status 1:
     command: /home/jeffery-hw/anaconda3/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6pqjycwt/message/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6pqjycwt/message/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-7pdeb4ce
         cwd: /tmp/pip-install-6pqjycwt/message/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-6pqjycwt/message/setup.py", line 6, in <module>
        import message
      File "/tmp/pip-install-6pqjycwt/message/message/__init__.py", line 10, in <module>
        ] + message.__all__ + observable.__all__
    AttributeError: partially initialized module 'message' has no attribute '__all__' (most likely due to a circular import)
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

Jeffery-Zhou, 2020-11-25:
I copied the old version from my deal colleague, it works now. Can't you believe that?

luismesas, 2021-03-05:
That exception is now fixed on version 1.3.0

closed 2021-03-05 by luismesas