luismesas/pydobot #35: Controlling the Gripper

pgsell, 2020-12-13:
Is there any way to control the gripper (open, close)?

I can only enable it, but not open and close or disable it again. Is this method missing or am I missing something?
Thank you for your help!

luismesas, 2021-03-05:
Hi to activate the gripper use the `.grip()` method with either `enable=True` to close or `enable=False` to open

To deactivate you need to deactivate the suction device, `.suck(False)`

It's not very intuitive, I'll rework the methods to they are more intuitive in a future version.

Does this clarifies how to use it?

danaswa, 2021-03-06:
I am having a problem, closing the gripper; the gripper will open, but not close.. 
Also I don't know how to turn off the air compressor after the code has finished. 
Thanks!

luismesas, 2021-03-07:
Is it a Magician Lite?