luismesas/pydobot #36: Clear the Alarm

SForney54, 2021-01-25:
Is there any way that can clear the alarm

Python code will not work unless resetting the Dobot Magician's current alarm

I don't think this library built-in with Alarm commands yet.

Thanks in advance

ZdenekM, 2021-02-12:
I have implemented reading alarms in my fork, see https://github.com/ZdenekM/pydobot/blob/master/pydobot/dobot.py. There are `get_alarms` and `clear_alarms`.

SForney54, 2021-02-12:
> I have implemented reading alarms in my fork, see https://github.com/ZdenekM/pydobot/blob/master/pydobot/dobot.py. There are `get_alarms` and `clear_alarms`.

ZdenekM, many thanks, I took a quick check in your fork. That alarm was quite helpful. I decided to download your code, used your example code and I get the result something like this, care to explain why I am getting this error message that the 'mode' isn't defined?

>>>%Run Example.py
Pose(position=Position(x=10.184884071350098, y=177.5911865234375, z=4.0564422607421875, r=86.7176742553711), joints=Joints(j1=86.7176742553711, j2=14.331169128417969, j3=59.5638313293457, j4=0.0))
Traceback (most recent call last):
  File "/home/pi/Desktop/OtherPyDobot/pydobot-master/alarm.py", line 12, in <module>
    device.move_to(position.x + 20, position.y, position.z, position.r, wait=False)
  File "/home/pi/Desktop/OtherPyDobot/pydobot-master/pydobot/dobot.py", line 589, in move_to
    return self._extract_cmd_index(self._set_ptp_cmd(x, y, z, r, mode))
NameError: name 'mode' is not defined
>>>

ZdenekM, 2021-02-16:
@SForney54 Which example code do you mean? I probably haven't touched https://github.com/ZdenekM/pydobot/blob/master/examples/basic.py at all. Try removing `wait=False`.

jKostet, 2021-12-06:
Hi, could this feature be merged to the main repo. Looks like move commands get sent to Dobot like normal, but the Dobot doesn't move if there are alarms about some axis being too close to limit.

jonascj, 2021-12-06:
@jKostet Absolutely, that is the plan. I am helping luismesas with issues and maintenance and I would like to do a new release this month with some small improvements, like clearing the alarm and not crashing on empty messages (#44). Maybe this is the most straightforward thing to fix, so the next release could just include this.

I still need to look at @Sforney54's work myself. At this point in time I'm not sure whether the library should try to clear axis alarms itself to keep the arm moving, or if it should raise an exception and offer functionality to clear the alarm and retry the move. Sforney54 seem to clear the alarm in the constructor, i.e. clear alarms on connection to the robot. This is sensible, but what about during operations?

jKostet, 2021-12-09:
Not sure but looking at the hardware I don't see physical end switches on the Dobot, so I think it just relies on absolute encoders, + the servo having a fourth cable for reading position. Thus I'd assume the limits are fixed in firmware and it should be safe Dobot-wise to actively ignore the alarms, e.g. having some variable that can be set in init like `device.ignoreAlarms = true`. Though this would then lead to the Dobot moving to unexpected positions environment-wise if a single move in a sequence of moves is not completed. Soo guess the expection + way to continue sequence after that would be best.

craigvear, 2022-09-13:
Apologies for jumping into this existing thread, after the fact. @ZdenekM I noticed that the command id for get_alarms and clear_alarms are both 20, yet the documentation, API and comms protocol for Dobot split them so that clear_alarms is 21. Is there a reason for your code using 20 for both? (PS I'm using Gustafsson's dobot-python repo, but the comms process is the same)