luismesas/pydobot #37: Pydobot Gripper Issue (Cannot Disable Gripper)

lukerbs, 2021-03-02:
Hi,

Is there currently a way to disable the gripper attachment once it has been activated? Currently I am only able to open and close the gripper, but not able to fully release. Thanks!

luismesas, 2021-03-05:
Do you mean disabling the suction to fully release pressure in the grip?

luismesas, 2021-03-05:
To release the pressure in the gripper you have to deactivate the suction device

```
device.suck(False)
```

Let me know if the trick works

lukerbs, 2021-03-05:
Instead of fully releasing pressure, **device.suck(False)** appears to activate the grip.
With **device.suck(True)** , the gripper is also still activated but in the reverse (it reverses the pressure to open the gripper).

The only way I am currently able to deactivate the pressure is by restarting the dobot via the physical on/off button.

luismesas, 2021-03-05:
When you do `device.suck(True)` does the grip closes?

`device.suck(False)` should deactivate the vacuum.

I have ran some tests on my end and if I do the following if opens the grip and then releases pressure.
```
device.grip(False)
device.wait(100)
device.suck(False)
```

lukerbs, 2021-03-05:
When I reset the dobot, the gripper arm is flaccid.

Running **device.suck(True)** causes the gripper attachment to open open up. (I can hear the air pump sucking).
Running **device.suck(False)** reverses the air pump and causes the gripper attachment to grip. (air pump reverses and stays on).

So **device.suck(False)** causes the air pump to continuously blow air out (instead of simply turning off the vacuum pump).

luismesas, 2021-03-05:
Ok looks like somehow your sucking device is working in a different way to mine.

luismesas, 2021-03-05:
can you confirm pydobot and firmware version?

luismesas, 2021-03-05:
Let's try one more thing, please execute this piece of code and reply with the log output

``` python
    device = pydobot.Dobot(port=port, verbose=True)

    device.wait(1000)
    device.grip(True)
    device.wait(1000)
    device.grip(False)
    device.wait(1000)
    device.suck(False)

    device.close()
```

lukerbs, 2021-03-05:
**Running the above code, the following occurs:**
**1.** Gripper closes
**2.** Gripper opens
**3.** Gripper closes again and stays closed 

I am using **pydobot version 1.3.2**
It may be worth noting that the Dobot model I am using is the Dobot Magician Lite.

luismesas, 2021-03-05:
Can you copy-paste here the console output with all the serial messages going back and forward?

lukerbs, 2021-03-06:
**Here is the console output:**

pydobot: /dev/ttyACM0 open
pydobot: >> AA AA:2:240:1::15
pydobot: << AA AA:2:240:1::15
pydobot: >> AA AA:2:245:1::10
pydobot: << AA AA:2:245:1::10
pydobot: >> AA AA:34:80:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:85
pydobot: << AA AA:10:80:3:DB 00 00 00 00 00 00 00:210
pydobot: >> AA AA:18:81:3:00 00 48 43 00 00 48 43 00 00 48 43 00 00 48 43:128
pydobot: << AA AA:10:81:3:DC 00 00 00 00 00 00 00:208
pydobot: >> AA AA:10:82:3:00 00 20 41 00 00 48 43:191
pydobot: << AA AA:10:82:3:DD 00 00 00 00 00 00 00:206
pydobot: >> AA AA:10:83:3:00 00 C8 42 00 00 C8 42:150
pydobot: << AA AA:10:83:3:DE 00 00 00 00 00 00 00:204
pydobot: >> AA AA:2:10:0::246
pydobot: << AA AA:34:10:0:54 4D 49 43 EF 03 D4 C1 01 EF 85 41 00 00 F0 C0 00 00 F0 C0 7A E4 8B 41 3D F2 65 42 00 00 00 00:44
pydobot: x:201.3                             y:-26.5                             z:16.7                             r:-7.5                             j1:-7.5                             j2:17.5                             j3:57.5                             j4:0.0
pydobot: >> AA AA:6:110:3:E8 03 00 00:164
pydobot: << AA AA:10:110:3:DF 00 00 00 00 00 00 00:176
pydobot: >> AA AA:4:63:3:01 01:188
pydobot: << AA AA:10:63:3:E0 00 00 00 00 00 00 00:222
pydobot: >> AA AA:6:110:3:E8 03 00 00:164
pydobot: << AA AA:10:110:3:E1 00 00 00 00 00 00 00:174
pydobot: >> AA AA:4:63:3:01 00:189
pydobot: << AA AA:10:63:3:E2 00 00 00 00 00 00 00:220
pydobot: >> AA AA:6:110:3:E8 03 00 00:164
pydobot: << AA AA:10:110:3:E3 00 00 00 00 00 00 00:172
pydobot: >> AA AA:4:62:3:01 00:190
pydobot: << AA AA:10:62:3:E4 00 00 00 00 00 00 00:219
pydobot: /dev/ttyACM0 closed

jobo89, 2021-05-13:
@lukerbs I am also using a DOBOT Magician Lite and was able to fix the suction cup issue.

Problem:
`bot.suck(True)`
Would start sucking
`bot.suck(False)`
Would cause the suction cup to blow. This should cause it to turn off

Solution:
in dobot.py:
change from:
...def _set_end_effector_gripper(self, enable=False):
......msg = Message()
......msg.id = CommunicationProtocolIDs.SET_GET_END_EFFECTOR_GRIPPER
......msg.ctrl = ControlValues.THREE
......msg.params = bytearray([])
......`msg.params.extend(bytearray([0x01]))`
......if enable is True:
.........msg.params.extend(bytearray([0x01]))
......else:
.........msg.params.extend(bytearray([0x00]))
......return self._send_command(msg)
to:
...def _set_end_effector_gripper(self, enable=False):
......msg = Message()
......msg.id = CommunicationProtocolIDs.SET_GET_END_EFFECTOR_GRIPPER
......msg.ctrl = ControlValues.THREE
......msg.params = bytearray([])
......if enable is True:
.........`msg.params.extend(bytearray([0x01]))`
.........msg.params.extend(bytearray([0x01]))
......else:
.........`msg.params.extend(bytearray([0x00]))`
.........msg.params.extend(bytearray([0x00]))
......return self._send_command(msg)

@luismesas I am not comfortable enough with git/pip to push this change myself but can you please update this?

I'll look into the gripper later tonight/tomorrow.

lukerbs, 2021-05-14:
@jobo89 Thanks for looking into this.

I suggest 3 gripper options:
**gripper open, gripper close, and gripper off**

jobo89, 2021-05-14:
@lukerbs sorry, that code should be for suck not grip. Also, I think you are right, 3 states would be good. I used 0, 1, and 2 but is there a way to pass an ENUM? I'm just getting back into python from C so I'm fighting adding a semicolen at the end of each line :)

Here is my dobot.py and Testing.py

Changes:
`_set_end_effector_gripper` now has 3 states
`_set_end_effector_suction_cup` now changes both bytes
[Archive.zip](https://github.com/luismesas/pydobot/files/6475907/Archive.zip)