luismesas/pydobot #44: Exception when response empty

Arzeg, 2021-09-08:
Hi,
After a few minutes with the Dobot Magician Lite I'm getting this exception, because of an empty response:
```
Traceback (most recent call last):
  File "direct_cords_read.py", line 81, in <module>
    device.move_to(current_x, y + draw_start_y, z_down, 0.0, wait=True)
  File "/home/pi/.local/lib/python3.7/site-packages/pydobot/dobot.py", line 315, in move_to
    self._set_ptp_cmd(x, y, z, r, mode=PTPMode.MOVL_XYZ, wait=wait)
  File "/home/pi/.local/lib/python3.7/site-packages/pydobot/dobot.py", line 240, in _set_ptp_cmd
    return self._send_command(msg, wait)
  File "/home/pi/.local/lib/python3.7/site-packages/pydobot/dobot.py", line 103, in _send_command
    expected_idx = struct.unpack_from('L', response.params, 0)[0]
AttributeError: 'NoneType' object has no attribute 'params'
```

I found this solution, and added the if clause in the middle:
```
    def _send_command(self, msg, wait=False):
        self.lock.acquire()
        self._send_message(msg)
        response = self._read_message()
        self.lock.release()
        if response is None:
            print("Receive empty response")
            return

        if not wait:
            return response
```

Best regards

jonascj, 2021-12-04:
@Arzeg Thank you for reporting this and thank you for your proposed solution. We will aim to fix this in the next release which will hopefully come during december. 

The solution is unfortnuately not as simple as your proposal. If the optional parameter `wait` is set to `True` then your solution skips the rest of the `_send_command()` function body. Other functions such as `pose()` relies on the message not being empty.

carlscho, 2025-09-15:
Any word on this / other ideas for fixes?