luismesas/pydobot #51: How to Set Jog Command ?

chinghiep01, 2022-12-06:
In the dType library , there is a set jog command where I could control jog by select the specific JOG . How could I add the set jog command in the library. I use the protocol and API user manual to set the function. The way I set similar to PTP command in pydobot library , except I change the protocol ID and control value to four .  As the result,  dobot not response  or running. Any advice to add that function in library ?

Prince25, 2023-03-03:
I have the same request. Any suggestions, @luismesas ?

chinghiep01, 2023-03-13:
for the set jog , you could use similar code as set ptp command. You need to add the protocol ID of set jog  and change the protocol ID in the code. You could find the protocol ID by use the document that luismesas attach at the description. Hope this help.

chinghiep01, 2023-03-13:
> I have the same request. Any suggestions, @luismesas ?

for the set jog , you could use similar code as set ptp command. You need to add the protocol ID of set jog and change the protocol ID in the code. You could find the protocol ID by use the document that luismesas attach at the description. Hope this help.